leon-vv/Traceon #206: Combine logic of solve_superposition and FieldSuperposition

jobdewitte, 2025-02-08:
## Problem statement

The `Solver.solve_direct_superposition` method returns a dictionary of fields. With the introduction of `FieldSuperposition` it seems more consistent to return a field superposition instead. The problem is that `FieldSuperposition` stores its underlying fields as a list rather than a dictionary of names and fields.

## Proposed Solution
Adapt `FieldSuperposition` by including a list of fieldnames. For the `solve_direct_superposition` method these can become the names of the underlying electrode (or `Component` see #205). This also helps to extract individual fields from a superposition more easily. 
Unnamed fields can still be supported by assigning `None` as name, so it wouldn't be a breaking change.

jobdewitte, 2025-02-08:
@leon-vv would it be an idea to add this to #204?

leon-vv, 2025-02-08:
I think this is a great idea, but I would prefer to take it up in a next PR

PR #239 (2025-03-19): Return FieldSuperposition from solve_direct_superposition
Closes #206
closed 2025-03-30 by leon-vv